amsbr/OctoPrint-EEPROM-Marlin #4: Add Z probe offset

brotherchris, 2017-04-25:
Hello, I find myself changing the Z probe offset a lot in Marlin. Can this be added to this plugin?
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0

Thanks,
Chris

amsbr, 2017-04-26:
Hi,

what's the firmware version?

brotherchris, 2017-04-26:
1.1.0-RC8

amsbr, 2017-04-28:
Same as me.

Current release of plugin have Z probe offset option for configuration. Check if it helps you

brotherchris, 2017-04-28:
Thanks I will give it a try.

ToonMeynen, 2017-11-16:
Hi,
I have set the offset several times. It was default 0, I changed it to -0.2 but when I hit upload, I get a communication timeout error. I need to reset ramps and reconnect octa. When I check the firmware settings again it shows 0 offset. It never actually saved.

amsbr, 2017-11-17:
Hi @ToonMeynen,

please try to send manual command at terminal M851 Z-0.2, verify the result.

amsbr, 2018-01-23:
Hi @ToonMeynen,

Did you verify the result?

averter, 2018-02-02:
Hi,

This plugin is a must have for people who want to tweak the firmware to their needs. Awesome work really.
I was wondering if it is possible to have the X and Y offset of the probe in addition to the Z offset? Thanks.

amsbr, 2018-02-04:
Hi @averter Marlin Firmware and this plugin support Home Offset for X, Y, Z and Z-Probe Offset.

averter, 2018-02-05:
Hi @amsbr Thanks for the quick reply. I think home offset and prove offset are different variables. I have Marlin RC6 with an autobed leveling probe. Do you think that in the future it would be possible to implement these three variables?
#define X_PROBE_OFFSET_FROM_EXTRUDER 39  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -34   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.0   //  Z offset: -below +above  [the nozzle]
I have to update them so many times that is such a tedious task. Your plugin looked as a salvation. Thanks in advance for any update.

amsbr, 2018-05-02:
Hi @averter,

XYZ probe offset from extruder currently isn't an EEPROM parameter. I'll suggest to Marlin firmware team to add this.

Thanks!

LMS0815, 2019-06-25:
I am using ABL Expert Plugin for this purpose. 
Is there any other Plugin available fpr Z probe offset adjustment?

geoffrothman, 2020-06-16:
> I am using _ABL Expert Plugin_ for this purpose.
> Is there any other Plugin available for Z probe offset adjustment?

@LMS0815 Yup, the EEPROM Marlin Editor plugin