amsbr/OctoPrint-EEPROM-Marlin #5: z jerk and e jerk mixed up

broncosis, 2017-05-04:
I went to adjust the jerk settings and they populat in the wrong window 
z jerk is shown in the e jerk and vise versa but oddly the changes made to seem to get stored in the correct location 
I am currently on 1.1.0 rc 2 I think 
but I don't know if this is a firmware problem

broncosis, 2017-05-04:
yes the E jerk is displayed in the Z window and Z is displayed in the E window
but changes are saved to the correct location I think

amsbr, 2017-05-05:
Hi @broncosis,

Thankyou for notify this issue. I will correct this and release a new version.

amsbr, 2017-08-20:
New release corrects this issue. Release [1.0.3](https://github.com/amsbr/OctoPrint-EEPROM-Marlin/releases/tag/1.0.3)

closed 2017-08-20 by amsbr