amsbr/OctoPrint-EEPROM-Marlin #7: ALL Jerk values are mixed up

azalanono, 2017-05-24:
Hello, Sylvia.
I am using "Marlin rc8" and want to inform you that **almost all Jerk values are wrong** in your EEPROM Editor. As Browser I use Firefox on Linux.

Details:
.) I click on "Load EEPROM"
.) I edit the Max Jerk values to Jerk X/Y/Z/E = 6/5/4/3   (-> as an example, to have different values).
.) I click on "Upload EEPROM". 
.) I click on "Load EEPROM" again (to read in what I configured before)

What a surprise, the jerk values are now showing up as 6/6/6/3  !!!!!

Now I change the settings in the LCD display of the printer, I set there the jerk for X/Y/Z/E to 6/6/2.9/5. After loading again the EEPROM in your editor, I see this horrible result:  6/6/6/2.9 !!!!!! 

Your software definitely reads in and/or parses received data incorrect!
Another explanation might be that you did too often a copy&paste when coding the edit fields for the jerk values and forgot hereby to change the variable names.

I remember very well that your older version of EEPROM Editor worked correctly.

This wrong behaviour makes your previously good plug-in more or less useless. I do not really trust it any more.

amsbr, 2017-05-29:
Hi @azalanono,

I'm writing a new version to correct this.

CyberDracula, 2017-08-15:
it is just a miss count on the parsing of M205 data. 
X and Y have the same match ID:

```javascript

                    self.eepromData2.push({
                        dataType: 'M205 X',
                        label: 'Max X jerk',
                        origValue: match[8],
                        value: match[8],
                        unit: 'mm/s',
                        description: ''
                    });

                    self.eepromData2.push({
                        dataType: 'M205 Y',
                        label: 'Max Y jerk',
                        origValue: match[8],
                        value: match[8],
                        unit: 'mm/s',
                        description: ''
                    });

                    self.eepromData2.push({
                        dataType: 'M205 Z',
                        label: 'Max Z jerk',
                        origValue: match[10],
                        value: match[10],
                        unit: 'mm/s',
                        description: ''
                    });

                    self.eepromData2.push({
                        dataType: 'M205 E',
                        label: 'Max E jerk',
                        origValue: match[12],
                        value: match[12],
                        unit: 'mm/s',
                        description: ''
                    });
                }
```

amsbr, 2017-08-20:
Hi @CyberDracula!

I am developing new release now.

Thanks!

amsbr, 2017-08-20:
New release corrects this issue. Release [1.0.3](https://github.com/amsbr/OctoPrint-EEPROM-Marlin/releases/tag/1.0.3)

closed 2017-08-20 by amsbr