amsbr/OctoPrint-EEPROM-Marlin #9: Not loading eeprom

storemanuk, 2017-10-10:
I've installed the plugin and see it listed with other plugins.

with printer connected I click the plugin from the left navbar then click on Load EEPROM but it doesn't list any of the eeprom settings. Just a blank page apart from a bit of text at the top and the two buttons
[image]

amsbr, 2017-10-15:
Hi @storemanuk 

The setting #define EEPROM_SETTINGS is enabled on your firmware?

storemanuk, 2017-10-16:
Duh! My apologies,

M503 was disabled in the firmware.
I recently uploaded Marlin 1.1.6 using a predefined configuration.h for the CR-10 but this had been disabled to allow other functions to fit in the program memory. I only discovered this last night and changed it. I'd rather have this feature than a boot screen so the boot screen went to give the room required for this feature. My newbie mistake.

amsbr, 2017-10-16:
Hi @storemanuk 

All right!

closed 2017-10-16 by amsbr