amsbr/OctoPrint-EEPROM-Marlin #10: Marlin 1.1.6 not loading EEPROM

InvalidExcepti0n, 2017-11-03:
After updating to Marlin 1.1.6 (CR-10S) pressing Load EEPROM does nothing
Not sure if is version issue or something else
If you need any logs, details let me know so I can provide

GitJiggyWiddit, 2017-11-04:
Also experiencing this problem on CR-10 running 1.1.6.

amsbr, 2017-11-05:
Hi @InvalidExcepti0n,

please send log details.

Thankyou

InvalidExcepti0n, 2017-11-05:
Hi @amsbr 

Will M503 will be enough? Or you need something else?

amsbr, 2017-11-05:
Only M503.

InvalidExcepti0n, 2017-11-05:
I have tried but not working M503 looks like need to enable debug output in firmware
Will post here when modified it later today
Sorry :(

InvalidExcepti0n, 2017-11-05:
Managed to get M503
so here is log from terminal:
`[...]
Send: M503
Recv: echo:M503
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E95.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z100 E25
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P5000.00 R1000.00 T5000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X30.00 Y30.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H200 B66 F0
Recv:   M145 S1 H250 B80 F0
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:  M304 P10.00 I0.02 D305.40
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.50
Recv: ok
[...]`

amsbr, 2017-11-06:
What's your version of OctoPrint?

InvalidExcepti0n, 2017-11-06:
Version 1.3.5 (master branch)

InvalidExcepti0n, 2017-11-07:
@amsbr  maybe this PR will solve issue on 1.1.6
https://github.com/MarlinFirmware/Marlin/pull/8283

when I get home will try it and post results

InvalidExcepti0n, 2017-11-07:
Working :)
offsets were bad that's why didn't work but now opened immediately after open
So closing issue and thx for support

closed 2017-11-07 by InvalidExcepti0n