amsbr/OctoPrint-EEPROM-Marlin #11: Trying to use editor with a Delta

imrcly, 2017-11-07:
I installed through the plugin manager repo running the current RC, the delta settings are not showing up when i load eeprom

amsbr, 2017-11-07:
Hi @imrcly 

Is it possible send M503 result?

imrcly, 2017-11-07:
[...]
Send: M503
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings:
Recv: echo:  M200 D1.75
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z80.00 E92.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z500.00 E100.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z1850 E5000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1000.00 R3000.00 T1000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X28.00 Y28.00 Z28.00 E5.00
Recv: echo:Endstop adjustment:
Recv: echo:  M666 X0.00 Y0.00 Z0.00
Recv: echo:Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
Recv: echo:  M665 L180.50 R100.00 H344.66 S200.00 B78.21 X0.00 Y0.00 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H200 B70 F255
Recv:   M145 S1 H240 B100 F255
Recv: echo:PID settings:
Recv: echo:  M301 P15.13 I0.95 D60.10
Recv: ok
[...]

imrcly, 2017-11-07:
forgot to mention Running marlin 1.1.5

amsbr, 2017-11-17:
Hi, @imrcly at the browser in the javascript console is firmware version printed something like "FIRMWARE_NAME: ...". Is it possible send this info?

Is none field showed after clicking in Load?

imrcly, 2017-11-17:
Send: M115
Recv: FIRMWARE_NAME:Marlin 1.1.5 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Tessa EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

i do not see a none field
[image]

amsbr, 2017-11-18:
Hi @imrcly,

I will prepare a new release for correct this.

amsbr, 2018-04-21:
Hi @imrcly,

fixed in the new release [1.2](https://github.com/amsbr/OctoPrint-EEPROM-Marlin/releases/tag/1.2)

closed 2018-04-21 by amsbr