amsbr/OctoPrint-EEPROM-Marlin #12: Plugin does not load values on Marlin 1.1.7

sanderpuh, 2018-01-06:


amsbr, 2018-01-19:
Hi @sanderpuh,

is it possible send M503 result?

amsbr, 2018-01-23:
Hi @sanderpuh,

your EEPROM setting is disabled on firmware compilation.

Wismgari, 2018-01-23:
Hi - just had the same issue with 1.1.8. EEPROM can be enabled and the plugin fail if the disable M503 option is used.

Ignore the warning in Marlin to disable M503 for release and the plugin works.

amsbr, 2018-01-23:
Hi,

yes, the plugin work only with M503 enabled. It uses the M503 result to populate the fields.

bcsteeve, 2018-02-02:
Man that was frustrating!  I struggled with this for hours before stumbling into here.  PLEASE consider making a check for this sort of thing, and having it suggest in the plugin to check that the necessary lines aren't commented out.  Perhaps also suggest to Marlin folks to make a note in the code documentation (eg. "this is necessary for some plugins, for example the EEPROM-Marlin" or whatever)

amsbr, 2018-02-04:
Hi @bcsteeve,

good suggestion!

amsbr, 2018-04-21:
Hi,

added support for all Marlin versions in new release [1.2](https://github.com/amsbr/OctoPrint-EEPROM-Marlin/releases/tag/1.2)

closed 2018-04-21 by amsbr