amsbr/OctoPrint-EEPROM-Marlin #13: Not recognize Marlin

Lepterois, 2018-01-30:
Hi! I have a Geeetech I3 printer with Marlin firmware, but the plugin say me, It's not a Marlin Firmware...
Do you have an idea?
Thanks

amsbr, 2018-02-02:
Hi @Lepterois 

Please execute M503 in Terminal and put here the result.

jsiobj, 2018-02-12:
Hello @amsbr,

I have the same problem with another printer : a Dagome Discoeasy 200 (a french manufacturer) which use an MKS board with a custom Marlin Firmware.

Here is the M503 output : 
```
Send: N70 M503*18
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z2560.00 E98.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z4.00 E170.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X9000 Y3000 Z100 E10000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P32.48 I6.40 D41.25 C100.00 L20
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.95
Recv: ok
```
Thanks.
JS

Lepterois, 2018-02-13:
Hi @amsbr !
I'm sorry for the late reply. I solved the problem by upgrading the firmware with a recent distrib of marlin. Almost everything is working fine, except the PID modification. When I try to change the setting of the PID, it saves only a blank field.
Thanks

TylonHH, 2018-03-24:
Can you please check this Version too?
```
Send: M115
Recv: FIRMWARE_NAME:Marlin 1.1.8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender 2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
```

amsbr, 2018-03-26:
Hi @TylonHH,

Please execute M503 in Terminal and put here the result.

TylonHH, 2018-04-02:
```
Send: M503
  G21    ; Units in mm
  M149 C ; Units in Celsius
Filament settings: Disabled
  M200 D1.75
  M200 D0
Steps per unit:
  M92 X80.00 Y80.00 Z400.00 E93.00
Maximum feedrates (units/s):
  M203 X500.00 Y500.00 Z5.00 E25.00
Maximum Acceleration (units/s2):
  M201 X500 Y500 Z100 E5000
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
  M204 P500.00 R500.00 T800.00
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
  M205 S0.00 T0.00 B40000 X20.00 Y20.00 Z0.40 E5.00
Home offset:
  M206 X0.00 Y0.00 Z0.00
Material heatup parameters:
  M145 S0 H205 B50 F0
  M145 S1 H245 B95 F0
PID settings:
  M301 P18.71 I1.41 D62.24
  M304 P252.00 I48.00 D326.00
Linear Advance:
  M900 K0.00 R0.00
ok
```

amsbr, 2018-04-03:
Hi @TylonHH,

I go to release a new version in some days for adjusting this.

Thank you.

amsbr, 2018-04-21:
Hi,

fixed in new release [1.2](https://github.com/amsbr/OctoPrint-EEPROM-Marlin/releases/tag/1.2)

try it.

closed 2018-04-21 by amsbr