amsbr/OctoPrint-EEPROM-Marlin #18: EEPROM GUI axis duplication

dsc45, 2018-04-25:
a lot of the settings are showing twice on my octoprint server.
[image]
[image]
[image]
[image]

Terminal Dump:
Send: M501
Recv: echo:V47 stored settings retrieved (626 bytes; crc 51397)
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 T1 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X66.99 Y66.99 Z1591.95 E144.96
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z7.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1200 Y1200 Z800 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P800.00 R1000.00 T1500.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X7.00 Y7.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Hotend offsets:
Recv: echo:  M218 T1 X0.00 Y27.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H180 B70 F0
Recv: echo:  M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.50
Recv: echo:Stepper driver current:
Recv: echo:  M906 X 800 Y 800 Z 800 E0 800 E1 800
Recv: echo:Sensorless homing threshold:
Recv: echo:  M914 X4 Y4
Recv: ok

amsbr, 2018-04-27:
has this occurred after load?

amsbr, 2018-05-01:
Yes. I observed this intermitance too. I'll fix it soon.

amsbr, 2018-05-02:
The cause of this problem related to dual or more simultaneous load (probably dual click or dual return of M501). I'll fix this soon.

Desterly, 2018-05-09:
I have this issue after modifying some values (just came to report) - for example, If Update my Z offset, when I click upload, I now have two boxes, one that shows the old offset, one that shows the new.  The issue is neither of them actually change the offset.   Only way to correct it is to reload the interface.     I haven't looked yet but perhaps a tuple or dictionary value to indicate that it's already been found/generated?

amsbr, 2018-05-09:
Hi @Desterly,

I'm checking this issue. Maybe it's related to a cleaning of a variable, I'll try a new commit soon.