amsbr/OctoPrint-EEPROM-Marlin #19: GUI does not respond

JupaCreations, 2018-04-27:
For your information...

Latest version loaded.
When I go to the GUI it does not show current settings.
When I click info button nothing happens.

Previous version worked fine.

[image]

ToonMeynen, 2018-04-29:
I believe it is browser a related issue. because in safari it works on the same installation.

amsbr, 2018-04-29:
thank you for the report. I'll test on other browsers

Hudefix, 2018-04-30:
Im having this issue on Safari and Yandex. Reinstall the plugin dont make a change!

amsbr, 2018-04-30:
Do you have some error message in javascript console?

JupaCreations, 2018-05-01:
IE gives same problem. Please describe where to find the javascript console.

amsbr, 2018-05-02:
Hi,

I tried another browser and not reproduce the problem. I identified an instability in autoload of settings from EEPROM, but after click on "Load" button (the "Info" isn't a button, is a tab, please click on "Load" for test), settings were loaded.

In my test, I used MacOS and Browsers:
- Safari 11.1
- Chrome 66.0.3359.139
- Opera 52.0
- Firefox 59.0.2
- Yandex 18.3.1.1235

I don't tried on Windows, I'll try soon.

amsbr, 2018-05-02:
For check javascript console in MacOS Browsers:

Command + Option + C: Safari
Command + Option + K: Firefox
Command + Option + J: Chrome
Command + Option + J: Opera
Command + Option + J: Yandex

JupaCreations, 2018-05-02:
I will check this evening which version of Chrome I use on my window system.
Clicking on other buttons does not give desired result either.
Auto load of settings at starting does not owrk.
I will make a video if needed for the behaviour.
It must be simple and as you say browser related.
Press Ctrl + Shift + J (Windows) for Java console

Paco

JupaCreations, 2018-05-02:
Here you go.....
https://paco.tinytake.com/sf/MjU2NDU1NV83NzEzNzQz

Paco

amsbr, 2018-05-02:
Hi Paco!

Probably that doesn't a browser compatibility problem. If you try to run in Terminal of OctoPrint the command M501 and M503, do you have any result?

JupaCreations, 2018-05-03:
Octorprint terminal works fine.... what else to check?

Paco
[image]

JupaCreations, 2018-05-15:
Can you make the previous version available as that worked fine? ;-)

JupaCreations, 2018-05-16:
For your information.
Installed the 1.0.4 version on two machines. Worked directly again............
What is the difference in use between old and new version?
Is there only cosmetic difference?

Thanks, Paco

Zance73, 2018-05-17:
It seems that I have the same problem
[image]
Octoprint 1.3.8 and EEPROM Marlin Editor Plugin: 1.2.1
Nothing special in the java console, and both M501 and M503 seems to work fine in the terminal.
Will try to downgrade to 1.0.4.