amsbr/OctoPrint-EEPROM-Marlin #27: Opening the Filament diameter page change the printer to E in MM3 automatically without saving or Upload

Xenomorphdelombre, 2018-07-14:
I installed the plugin and wanted to check all options. After looking around I was satisfied with my changed I saved and uploaded and tried a print which got Under Extruded.

Went back to all options and found the Filament tab which was set to 1.75. I changed it to 1 and my prints were somewhat "ok". Then I found online that this is to enable E in mm3 instead of linear extrusion.

It owuld be good if you'd be reading the setting and showing which is setup in the firmware and the give a checkbox to enable or not, the if enable set filament to whatever it is, with a more detailed explanation on what is E in mm3.

Thanks!

kbuechl, 2018-07-17:
+1 to this. I downloaded this plugin at the recommendation of a friend and since then my prints have been totally fubar. Came here and saw this issue and checked on my printer and it was turned on without me knowing.

amsbr, 2018-07-27:
Hi @Xenomorphdelombre, in the next release this feature to be released. @kbuechl sorry for the setback.