amsbr/OctoPrint-EEPROM-Marlin #28: Not Loading Data On SkyNet3D variant of Marlin

Quintin9801, 2018-07-16:
I've used this plugin on my printer running SkyNet3D before but after a few months of not using it I've come back to see it no longer loading my printers settings. I don't have any error messages appear on the screen and my "Info" button is greyed-out.

Pringer Info from plugin page:
FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/ErikZalm/Marlin/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:SkyNet3D EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
[image]

amsbr, 2018-07-27:
Hi @Quintin9801, the EEPROM feature was enabled on this firmware?

Quintin9801, 2018-07-27:
Problem solved , reflagged with latest version of skynet. No worries,
perfectly fine to close the issue

closed 2019-08-24 by amsbr