amsbr/OctoPrint-EEPROM-Marlin #29: Doesn't save settings

Tronnic, 2018-08-03:
**Issue:**
Hello, I need to change EEProm Settings in order to calibrate my extruder. When entering the number and hitting upload, I get a green message "EEPROM DATA STORED" but right after uploading the number I entered will revert back to the original number.

**Info:**
Printer: Creality Ender 3
Octoprint: 1.3.8
Octopi: 0.15.1 (Jessie light Octopi Image with TouchUI installed)

Tronnic, 2018-08-04:
This is from the console right after I change a setting (just to test, changed Extruder steps from 93 to 94.

`Send: M105
Recv: ok T:28.1 /0.0 B:28.9 /0.0 T0:28.1 /0.0 @:0 B@:0
Send: M92 E94
Recv: ok
Send: M500
Recv: ok
Send: M504
Recv: ok
Send: M504
Recv: ok
Send: M501
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E93.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X500 Y500 Z100 E5000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S500.00 T500.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P21.73 I1.54 D76.55
Recv: ok
Send: M105
Recv: ok T:28.0 /0.0 B:28.8 /0.0 T0:28.0 /0.0 @:0 B@:0`

edmundchaniii, 2018-08-07:
I just experienced this exact same issue with a Creality Ender 2.

edmundchaniii, 2018-08-07:
Oh, one other thing.   This may have something to do with it.

When you Load the EEProm settings and go to the "Steps per mm" Tab, it repeats the same "Steps" info to the screen 2 times instead of "Steps" and "Feedrates" like in the screen shot here:

https://plugins.octoprint.org/plugins/eeprom_marlin/