amsbr/OctoPrint-EEPROM-Marlin #30: Plugin restoring stored EEPROM settings with M501 when OctoPrint settings are viewed

tophattwaffle, 2018-08-21:
When viewing the settings for OctoPrint, the EEPROM editor plugin sends M501 to the printer. This will restore the stored EEPROM to be live. However this is not ideal as some settings may be set on a per-print basis.

Example: Automatic Bed Leveling only after G29. After a G29, Marlin will set M420 S1, enabling bed leveling. In my EEPROM I have M420 S0 set so ABL is only enabled after a G29 has generated the mesh. Opening my OctoPrint settings results in M501 being sent to my printer, setting M420 S0, disabling ABL - Along with whatever other print specific settings I may have had set to override EEPROM values.

I don't think the plugin should be restoring settings just for viewing it's settings. I'm guessing you've used M501 to get the settings into terminal so you can parse them out for the view the stored settings. If there was some sort of warning to let people know loading settings may cause issues, that would be nice. Or maybe a toggle to switch between M501 (stored) and M503 (live) settings.

This was a fun few weeks of troubleshooting to where I have now removed this plugin as it has been killing my prints.

Desterly, 2018-08-21:
The problem I see with this is there is no current command in Marlin to "View" EEPROM Data only:

M500 - Save to EEPROM
M501 - Load from EEPROM
M502 - Reset EEPROM
M503 - View Memory

M501 is no different than turning the printer off and back on.    

Are you saying there are times that you are changing printer settings but not saving them to EEPROM for multiple jobs?  

I'm also not sure how this affects ABL overall.   ABL is disabled every G28 and is only enabled with a G29 and/or M420 S1 

    M420 S can be used to enable/disable bed leveling. 
    For example, M420 S1 must be used after M501 to enable the loaded mesh or matrix, 
    and to re-enable leveling after G28, which disables leveling compensation.

What we ultimately need though is a M505 - View EEPROM data.    The only alternative that I could see would be if the plugin did a M503 to bring in the settings, M501 to load, then after a "save" or "close" of the editor, restored the values from the M503.

Or am I missing something else?

tophattwaffle, 2018-08-21:
You are correct, a M505 would solve all of this. I'm going to make a
request over at Marlin to see if this is possible.

It effects ABL for many users as M420 S0 is set in the eeprom. The starting
script then contains a G28 (home and clear old mesh) followed by G29 which
makes a new mesh and saves it.

The issue is that if you're in the middle of a print and ABL is on, this
plug-in does a M501 and it loads the M420 S0 that ends up disabling ABL.

Desterly, 2018-08-22:
While I'm not sure exactly what difference enabling/disabling bed leveling (unless you are also referring to Z compensation) has during printing, based off your updated description, what you are looking for is prevention of clicking "Load" and issuing the M501 if printing is active. 

This would prevent the M501 being issued during an active job.  G28 always disables bed leveling which means in startup scripts, a G29 or M420 S1 MUST be issued to enable bed leveling anyway.  

As a side note, G29 does not save the new mesh.  unless I'm drastically mistaken, only a M500 will actually store the new mesh to eeprom.

Ideally though, I think a M500-M504 should all be prevented/disabled during an active job

tophattwaffle, 2018-08-22:
Yes, bed leveling as in z compensation.

And yes - if a print is active a M501 should never be issued. Thought
preventing M500-504 during active print also works.

With EZABL a G29 is issued every print and the mesh is never saved.

Desterly, 2018-08-22:
OK overall I see the issue...     

This is actually a fairly easy fix as I recall (I've looked at the plugin a few times).    I'll see what I can do and issue a pull request to disable the admin functions during a print (if amsbr doesn't get to it first)

Desterly, 2018-08-23:
Take a look at pull request #32 -  This is a small modification that effectively disables the Add-In if the printer is currently "Printing" or "Paused"