amsbr/OctoPrint-EEPROM-Marlin #41: Only info tab shows for 1.1.9 Bugfix

fcollingwood, 2019-01-23:
None of the settings tabs show for 1.1.9 Bugfix

Is this being maintained? There's an awful lot of open issues and bugs that date back quite some time
[image]

JBFUK, 2019-02-27:
Same issue with Marlin 1.1.8; perhaps it's the OctoPrint version?

Akegata, 2019-03-04:
I have the same issue in the 2.0 bugfix release of Marlin, so OctoPrint version sounds like a likely commonality.
I'm running 1.3.10.

Akegata, 2019-03-04:
If you still have an issue with this, I apparently "fixed" it for me by changing the Z probe offset in the terminal (M851 Z-1.2), that somehow triggered something that made the EEPROM plugin work again.

barnold96, 2019-05-09:
Interesting!  I've been having a few issues and also ran into this today.  After making changes to a section of the configuration.h file, One error went away, but this one arose.  I changed the following:
//
#define EEPROM_SETTINGS // Enable for M500 and M501 commands  //BA_CHANGE
//#define DISABLE_M503    // Saves ~2700 bytes of PROGMEM. Disable for release!
#define EEPROM_CHITCHAT   // Give feedback on EEPROM commands. Disable to save PROGMEM.  //BA_CHANGE

//
// Host Keepalive
//
// When enabled Marlin will send a busy status message to the host
// every couple of seconds when it can't accept commands.
//
//#define HOST_KEEPALIVE_FEATURE        // Disable this if your host doesn't like keepalive messages  //BA_CHANGE
//#define DEFAULT_KEEPALIVE_INTERVAL 2  // Number of seconds between "busy" messages. Set with M113.  //BA_CHANGE
//#define BUSY_WHILE_HEATING            // Some hosts require "busy" messages even during heating  //BA_CHANGE

When I attempted to check the eeprom settings, I, too, had only the INFO button.  Just for the heck of it, I clicked the LOAD button and nothing happened,  Then, I clicked the UPLOAD button and all the headings appeared!  Don't ask me how - just reporting what I saw.

thoscut, 2019-05-16:
Disabling EEPROM_CHITCHAT can cause this issue, as Marlin doesn't return a request result as expected by the plugin.

leftover, 2019-05-24:
Sequence: M501 M500 worked for me.
(EEPROM reset)

TarruTarru, 2019-12-01:
Hello guys, 

I have now exactly these same problem. With Marlin 2.0.X from June - everything works perfectly fine. But when I upgrade Marlin to the newest version available on GitHub - I don't have "Advanced" tab in EEPROM editor. 

Do You know why this happens? When I am getting back to this version from June - "advanced" tab (and every other tab) are there. 

Thanks for help.

SVLoneStar, 2020-01-06:
@TarruTarru I have the same issue and after looking into the code of the EEPROM plugin it seems that the format for certain messages was changed within Marlin, so that the plugin does no longer recognize them.

I believe the issue to be in line 23 of eeprom_marlin.js (which, from its code comments, is dated from July 2015). There it says

`self.eepromM851RegEx = /M851 ([Z])(.*)/;`

I replaced this by 

`self.eepromM851RegEx = /(M851.*)([Z])(.*)/;`

I checked this in https://regex101.com/, and it seems to match the Z offset correctly.
After downloading the plugin's ZIP, Imodified the file
OctoPrint-EEPROM-Marlin-master\octoprint_eeprom_marlin\static\js\eeprom_marlin.js (as shown above) and re-installed the plugin. Now, the Advanced tab is visible, but the value is not (yet) filled...digging deeper...
Any ideas welcome...