amsbr/OctoPrint-EEPROM-Marlin #50: HTML page refresh triggering M503 even when printing in progress

gruvin, 2020-04-05:
Marlin 2.0 is my only option for my 32-bit board. Marlin 2.0 (including bugfix version) is riddled with bugs. One of those bugs is that issuing an M503 during printing will mess up motion planning somehow, resulting in unintended X,Y or Z stepper kitters, which in turn almost always destroys whatever is printing. (It doesn't *always* happen and was thus hard to track down. But it does.)

Now, none of that shouldn't be your concern HOWEVER ...

It would appear the this plugin is somehow triggering an M503 to be sent to the print upon page load or reload. A fresh load on my laptop, having left mydesktop will do this.

IMO, this plugin should NEVER issue an M503 or any other GCODE until it is activated and on screen, in the settings menu.

Thanks for you time, work and consideration. You are part of "the solution" on this planet and that is very much appreciated.

gruvin, 2021-09-09:
Well that got nowhere.

closed 2021-09-09 by gruvin

cp2004, 2021-09-09:
I took over development - if you have issues, feel free to raise them at https://github.com/cp2004/OctoPrint-EEPROM-Marlin

That said, I don't think this is an issue any more.