amsbr/OctoPrint-EEPROM-Marlin #61: Is this plugin still being maintained? Adoption request

cp2004, 2020-09-22:
Hey @amsbr!
Is this plugin still being maintained?

If I don't hear back from you, I will look into adopting it on the plugin repository and become the new maintainer of this plugin.

It needs things like Python 3 compatibility, and a few bug fixes.

I think it has some great potential, with over 5k confirmed installs and some promising ideas.

I will look at creating a fork in the coming days, or if anyone else volunteers I am happy to collaborate, or even hand over to them.

cp2004, 2020-09-24:
For anyone who comes across it: I have forked this plugin, when it is two weeks since this issue was submitted I will take over this plugin. There are things I am changing, but it will definitely be an improved plugin. 

You can see my development ideas here:
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/issues/1

carl1961, 2020-09-28:
good question
I did a fork , install the plugin with the web address 
https://github.com/carl1961/OctoPrint-EEPROM-Marlin/archive/master.zip

cp2004, 2020-09-28:
Hey @carl1961!
I have forked it too, (link in my previous comment) but I would not recommend installing it since I still need to change the update hook. I have added Python 3 compatibility, and a couple of other bits too. 

Once I get some time (working on other plugins at the moment) and when it is 2 weeks from this issue, I will be able to take over the listing on the official plugin repository. 

I have some (probably ambitious) plans for it, if you want to contribute to my fork feel free!

carl1961, 2020-09-28:
Thanks @cp2004  I just realized it was your    python3 upgrade.py I used to convert from 2.7 to 3.7  works great. I been for a week trying to get OctoPrint running on a Orange pi lite,  and doing things with python3 stuff just cause issues. but installing the python2  octoprint install and then converting with your script to python3 did the trick. Thanks!

cp2004, 2020-10-05:
@amsbr In a couple of days I will be taking over the listing of the plugin repository, to point to my fork, since it seems this plugin has been abandoned.

If you want to change any of this (or transfer ownership of the repository, makes life a *lot* easier) then please get in touch as soon as possible.

cp2004, 2020-10-07:
The official repository listing has been transferred to my fork, since @amsbr is no longer maintaining it.

closed 2020-10-07 by cp2004